pschatzmann/arduino-espeak-ng #1: compile error in Plaftormio :

deladriere, 2022-11-12:
I am trying to compile your example with Platformio but I get many errors
like : 

```
from .pio/libdeps/esp32dev/espeak-ng/src/pcaudiolib/audio_audiotools_i2s.cpp:1:
.pio/libdeps/esp32dev/Posix Filesystem/src/FileSystems/Registry.h:122:17: error: 'file_systems::NoRegEntry' declared as an 'inline' variable
 inline RegEntry NoRegEntry;
                 ^
.pio/libdeps/esp32dev/Posix Filesystem/src/FileSystems/Registry.h:124:31: error: 'file_systems::NoFileSystem' declared as an 'inline' variable
 inline FileSystem NoFileSystem("/null");
                               ^
.pio/libdeps/esp32dev/Posix Filesystem/src/FileSystems/Registry.h:126:27: error: 'file_systems::open_files' declared as an 'inline' variable
 inline Vector<RegEntry *> open_files;
                           ^
.pio/libdeps/esp32dev/Posix Filesystem/src/FileSystems/Registry.h:128:29: error: 'file_systems::file_systems' declared as an 'inline' variable
 inline Vector<FileSystem *> file_systems;

```

here is my platformio.ini (it works fine with your Flite library)

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
board_build.f_cpu = 240000000L
framework = arduino
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder
monitor_raw = yes
board_build.partitions = partitions.csv

upload_port = /dev/cu.usbserial-0001

lib_deps =
   https://github.com/pschatzmann/arduino-espeak-ng
   https://github.com/pschatzmann/arduino-audio-tools
   https://github.com/pschatzmann/arduino-posix-fs

board_upload.maximum_size = 3145728
board_upload.flash_size = 4MB

;build_flags = -DCORE_DEBUG_LEVEL=0    ; None
;build_flags = -DCORE_DEBUG_LEVEL=1    ; Error
;build_flags = -DCORE_DEBUG_LEVEL=2    ; Warn
;build_flags = -DCORE_DEBUG_LEVEL=4    ; Debug
;build_flags = -DCORE_DEBUG_LEVEL=5    ; Verbose
build_flags =
        -DCORE_DEBUG_LEVEL=3            ; Info
        -DCONFIG_ARDUHAL_LOG_COLORS=1

```
thanks  !

pschatzmann, 2022-11-12:
Please define the platform as described in the documentation: https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO

deladriere, 2022-11-12:
Thanks ! 
I compile but I have a memory issue now
```
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  36.7% (used 120376 bytes from 327680 bytes)
Flash: [==========]  121.6% (useError: The program size (1594053 bytes) is greater than maximum allowed (1310720 bytes)
*** [checkprogsize] Explicit exit, status 1
d 1594053 bytes from 1310720 bytes)
==================================================================== [FAILED] Took 53.18 seconds ====================================================================

```

pschatzmann, 2022-11-12:
You need to set the partition scheme to Huge App

deladriere, 2022-11-12:
aha it works !!! thanks
but I have a glitch sound at the end of the sentence: https://www.dropbox.com/t/uZwqWdtaT35Eztzs

pschatzmann, 2022-11-12:
Can you share your platformio.ini, so that I can add it to the documentation in the Wiki

deladriere, 2022-11-12:
sure, you also need the following CSV file too (I got this somewhere but I don't understand what I am doing ...)
```
[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_deps =
 https://github.com/pschatzmann/arduino-audio-tools
 https://github.com/pschatzmann/arduino-espeak-ng
 https://github.com/pschatzmann/arduino-audio-tools
 https://github.com/pschatzmann/arduino-posix-fs
build_flags = -DCORE_DEBUG_LEVEL=2 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
monitor_speed = 115200
monitor_filters = esp32_exception_decoder

board_build.partitions = partitions.csv
```
and the partitions.csv file to store along with platformio.ini

```
# Espressif ESP32 Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x2F0000,
```

pschatzmann, 2022-11-12:
You should be able to use the built in partition tables.
Can you double check if the following works as well:
board_build.partitions = huge_app.csv

deladriere, 2022-11-12:
Confirmed : it works for me, thanks
Do you also have the glitch at the end of the sentence ?

pschatzmann, 2022-11-12:
Yes, I noticed this as well...

deladriere, 2022-11-12:
Can we set the speech rate and the pitch ?

pschatzmann, 2022-11-12:
I would expect so, but I have not tried yet...

pschatzmann, 2022-11-18:
The glitch at the end was only on the ESP32. I complied the sketch on the desktop and there it was just perfect.
I could resolve it by adding some silence at the end...

deladriere, 2022-11-18:
It reminds me of the discussion we had before  [thread](https://github.com/pschatzmann/arduino-SAM/issues/1)

pschatzmann, 2022-11-18:
I added some methods e.g. to define the pitch. For setting the voice I need some help: https://github.com/espeak-ng/espeak-ng/discussions/1542

closed 2022-11-23 by pschatzmann