pschatzmann/arduino-espeak-ng #2: Concatenation of letters fails for German umlaut characters

Pummelo65, 2022-12-28:
Building up on the provided example for using different voices I tried to include umlauts like ä, ü and ö in the spoken text.
Running this on an ESP32 it appears as if these unicode umlauts are treated as separate words i.e. single letters. Running it using the desktop version (cmake example) works as expected.
I checked if the additional break logic in FindReplacementChars causes issues with multibyte characters but removing it and compiling the project on Platform-io reveals the same behavior.
If other can reproduce this behavior I'ld be willing to dive deeper into it.
/Roland

pschatzmann, 2022-12-29:
The espeak_Synth() method accepts some flags to specify the input text type. I think I am using espeakCHARS_AUTO by default.
I suggest that you have a closer look if any change there helps.

Pummelo65, 2022-12-29:
Hi, I already tried forcing the input character set to be UTF8
unsigned int  flags = espeakCHARS_UTF8;

But without success.

I assume the culprit somewhere in the translation functions breaking words where the shouldn’t.

The Umlauts are correctly spoken out, just separated as if they were single letters between words.

I wonder if this just a particular Arduino setup - I am using platform-io for development.

Could you try a sentence like:

espeak.say(„Übermäßiger Genuss von Süßigkeiten macht träge.„);

With your setup and check if words get split?

Thanks and cheers

Roland

pschatzmann, 2022-12-29:
Yes it sounds absolutely crazy. I tried "schöne neue Welt" and it starts with spelling out the letters...

Pummelo65, 2022-12-29:
The strange thing is that if I compile it for desktop (espeak-cmake) it works well.
The only difference I can see aside from the output path is the different file handling for all the configuration tables.
I will start to println intermediate stuff to compare where the concatenation gets broken.

Pummelo65, 2022-12-30:
I found the root cause for the problem which is a combination of two issues.
The built-in implementation of iswalpha() in the esp32 tooolchain appears to be broken in the sense that it does not treat multibyte characters as letters.
In the espeak common.c ucd.h is included but the explicit ucd_ functions are not used.
The attached version solves the problem, may be it is time for another patch :-(
￼

pschatzmann, 2022-12-30:
I don't see any attached version.
will you be able to provide a pull request ?

pschatzmann, 2022-12-30:
Cool. your pull request made a huge difference! 
Thanks a lot

closed 2022-12-30 by pschatzmann