pschatzmann/arduino-espeak-ng #6: Words beginning with "st" or "ff" speak incorrectly?

pinter75, 2023-03-06:
Hi,

Great library, have it working on an ESP32 with an external I2S amp. All is good except with words beginning with "st".

S on its own speaks correctly, but any word beginning with "st" speaks out code.

In the example change 

espeak.say("Hello world!"); to espeak.say("Stuff"); to reproduce.

s 
audio_object::write(10734)- OK

stuff 
audio_object::write(22052) -?
audio_object::write(22052) -?
audio_object::write(22052) -?
audio_object::write(22052) -?
audio_object::write(22052) -?
audio_object::write(22052) -?
audio_object::write(3414) -OK

Also "ff" has the same behavior.

pinter75, 2023-03-07:
Ok this is working but odd..

  espeak.add("/mem/data/voices/!v/**Alex**", espeak_ng_data_voices__v_benjamin, espeak_ng_data_voices__v_benjamin_len);
  // setup espeak
  espeak.begin();
  // Set voice and voice option
  espeak.setVoice("en+**benjamin**");

Causes Warning: open: file '/data/lang/!v/benjamin' does not exist but fixes the st and ff issue!

pschatzmann, 2023-03-08:
The correct syntax to register benjamin would be 
```
  espeak.add("/mem/data/voices/!v/benjamin", espeak_ng_data_voices__v_benjamin, espeak_ng_data_voices__v_benjamin_len);
  espeak.setVoice("en+benjamin");
```
And you are right, benjamin, seems to work fine...

pinter75, 2023-03-08:
Yup confirmed Benjamin does work. No Idea why it still worked if i specced benjamin but used Alex. Anyways all is good!

Is there a way to slow down the voice, or is there a preset voice that is a bit slower?

  espeak.setRate(80);
  espeak.setWordGap(50);

Tried but the TTS can rattle through the words in a sentence quickly. It's almost like I need a slightly slower voice...

Also great work on porting this to the ESP, it's by far the best TTS I have heard on a low cost MCU!

pschatzmann, 2023-03-08:
I don't remember. Just look at the class documentation.
It seems that if no voice is loaded, no all sounds are supported, so any voice will actually work!

closed 2023-03-08 by pschatzmann