pschatzmann/arduino-espeak-ng #7: Included WIFI exceeds the memory limits

fenyvesi, 2023-05-25:
This is a really nice implementation.

I would like to use it to read loudly the error messages sent through MQTT.

The problem when I include just the WIFI connection, the dram0_0_seg overflowed by 12992 bytes.

Is there any potential work-around it?

Thank you.

Arduino v.1.8.19, WIN10

`c:/users/x/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\x\AppData\Local\Temp\arduino_build_285595/reader.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/users/x/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/x/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/users/x/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 12992 bytes
collect2.exe: error: ld returned 1 exit status
`

fenyvesi, 2023-05-25:
Yes.

pschatzmann, 2023-05-25:
If you have processor with psram you can try to use psram
https://www.pschatzmann.ch/home/2022/05/30/esp32-and-psram/

fenyvesi, 2023-05-25:
Thank you, I look around maybe I have it. Coming back.

fenyvesi, 2023-06-07:
I just got the ordered module with PSRAM and tried again in platformio. The problem is the same:
:/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 13104 bytes

I included the proposed heap_caps_malloc_extmem_enable() function but as it has effect only at runtime it has no benefit at compiling time.

Maybe I misunderstood you.
The code:

`#define WIFI_EXIST
#include <WiFi.h>
#include "esp_heap_caps.h"
#include "esp_system.h" //This inclusion configures the peripherals in the ESP system.
#include "esp32-hal-psram.h"
//#include "esp_himem.h"
#include "esp32/himem.h"

#include "AudioTools.h" // https://github.com/pschatzmann/arduino-audio-tools
//#include "AudioLibs/AudioKit.h" // https://github.com/pschatzmann/arduino-audiokit
#include "FileSystems.h" // https://github.com/pschatzmann/arduino-posix-fs
#include "espeak.h"

//const int limit = 8192;
const int limit = 4096;		// the minimum
I2SStream i2s; // or replace with AudioKitStream for AudioKit
const bool load_english = false;
ESpeak espeak(i2s, load_english);

void setup() {
	heap_caps_malloc_extmem_enable(limit);
	Serial.begin(115200);			//Initialize serial ports for communication with the console
	Serial.println("Program started");
	psramInit();
	vTaskDelay(3);
    Serial.printf("Total heap: %u\r\n", ESP.getHeapSize());
    Serial.printf("Free heap: %u\r\n", ESP.getFreeHeap());
	Serial.printf("Total PSRAM: %u\r\n", ESP.getPsramSize());
	Serial.printf("Free PSRAM: %d\r\n", ESP.getFreePsram());

  // add foreign language configuration files
  espeak.add("/mem/data/hu_dict", espeak_ng_data_hu_dict,espeak_ng_data_hu_dict_len);
//  espeak.add("/mem/data/lang/de", espeak_ng_data_lang_gmw_de, espeak_ng_data_lang_gmw_de_len);
  espeak.add("/mem/data/lang/hu", espeak_ng_data_lang_urj_hu, espeak_ng_data_lang_urj_hu_len);

  // setup espeak
  espeak.begin();
  espeak.setVoice("hu");

  // setup output
  auto espeak_info = espeak.audioInfo();
  auto cfg = i2s.defaultConfig();
  cfg.channels = espeak_info.channels; // 1
  cfg.sample_rate = espeak_info.sample_rate; // 22050
  cfg.bits_per_sample = espeak_info.bits_per_sample; // 16
  i2s.begin(cfg);

#ifdef WIFI_EXIST
    WiFi.begin("AP_NAME", "AP_PASS");
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
#endif
}
void loop() {
  espeak.say("Csöngettek!");
  delay(1000);
}

platformio.ini:

> [platformio]
description = Espeak-NG Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_deps =
    https://github.com/pschatzmann/arduino-audio-tools
    https://github.com/pschatzmann/arduino-espeak-ng
    https://github.com/pschatzmann/arduino-posix-fs
build_flags = -DCORE_DEBUG_LEVEL=5 
                -Wno-unused-variable 
                -Wno-unused-but-set-variable 
                -Wno-unused-function 
                -Wno-format-extra-args
                -DBOARD_HAS_PSRAM
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
board_build.partitions = huge_app.csv`

pschatzmann, 2023-06-07:
It should be possible to change some big arrays to be allocated dynamically.

pschatzmann, 2023-06-07:
You can try to activate the ESPEAK_HEAP_HACK and check if it is working with psram

fenyvesi, 2023-06-08:
Really nice touch. 

I used the last version and it works with MQTT messages. As I see ~80k is used from PSRAM. (	heap_caps_malloc_extmem_enable(4096);)

Thank you once more.

pschatzmann, 2023-06-08:
I was just running a test with psram and Wifi and it was working like a charm:

10:40:58.479 -> Total heap: 269996
10:40:58.479 -> Free heap: 183392
10:40:58.479 -> Total PSRAM: 4191915
10:40:58.479 -> Free PSRAM: 4122023

pschatzmann, 2023-06-08:
https://github.com/pschatzmann/arduino-espeak-ng/wiki/Using-PSRAM

fenyvesi, 2023-06-08:
It is useful for anybody who wants to use it. The usage of PSRAM can be problematic because of the common cache with the normal RAM. 

Just a remark: as far as I remember the minimum and default allocation is 4096 in the heap_caps...().

This is a really useful tool. For example if you would like to know your debug messages but there is no serial connection, then you can listen to it. For blind people you can construct a web page reader and so on.

As the vast majority of the PSRAM is not used, there are real possibilities.