pschatzmann/arduino-espeak-ng #8: espeak on esp32s2

eroboticdude, 2023-11-16:
Using the latest version of the esp32 drivers (3.0.0-a2), I get a compilation error:
```
c:/users/ellio/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\ellio\AppData\Local\Temp\arduino\sketches\E35BE471FBF17A680B0A4559C3979742\libraries\FS\vfs_api.cpp.o:(.literal._ZN11VFSFileImplD2Ev+0x4): undefined reference to `vtable for fs::FileImpl'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\ellio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha2\libraries\WiFi
  Not used: C:\Users\ellio\Documents\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
  Used: C:\Users\ellio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha2\libraries\SD
  Not used: C:\Users\ellio\AppData\Local\Arduino15\libraries\SD
  Not used: C:\Users\ellio\Documents\Arduino\libraries\SD
exit status 1

Compilation error: exit status 1
```

I tried to downgrade the drivers, but that prevented the AnalogAudioStream from being used.

pschatzmann, 2023-11-17:
I tested with espeak-arduino.ino example and could not reproduce your error.

In any case if you double check your error message: this is not even coming from this project!

closed 2023-11-17 by pschatzmann

eroboticdude, 2023-11-17:
> I tested with espeak-arduino.ino example and could not reproduce your error.
> 
> In any case if you double check your error message: this is not even coming from this project! I suggest that you uninstall and reinstall the ESP32 3.3.0-alpha2 kernal again...

Hi, did you test compiling it as esp32-s2? I think that's where that error came from. I will try to reinstall again.

pschatzmann, 2023-11-17:
Yes, I did...

eroboticdude, 2023-11-17:
ok sorry for the trouble, it worked for me after some restarting and recompiling. Thanks for your help!

eroboticdude, 2023-11-17:
I continue to get this same error: `undefined reference to 'vtable for fs::FileImpl`

pschatzmann, 2023-11-17:
I suggest to report this here https://github.com/espressif/arduino-esp32

eroboticdude, 2023-11-17:
It seems like an issue with Arduino IDE 2.2.1, it compiled fine on arduino 1.8.13